Bastardkb/Dilemma #3: Dilemma integrated review

KarlK90, 2022-07-08:
@bstiq Pretty much the same feedback as on the splintegrated - minus the ideal diode stuff as the dilemma doesn't feature RGB

# Recommendations

1. ESD protection for USB-C port and split-comms port on data lines only or including cc lines (like shown)
	- protects against static discharge (charged cloths!) and subsequent destruction of any IC
	- recommendation StMicroelectronics (or knock-off manufacturer) **USBLC6-2** in either SOT23-6L *C7519* or the smaller SOT-666 *C15999* package or the **USBLC6-4** as shown

[image]

2. Support full-duplex serial on the TRRS jack
	- much higher baudrates are possible, which in turn shortens the time to send data across the halfs. Thus frees up processing power
	- the kids want it :P

3. Expose SWD debugging headers, great for devs like me :-). The following signals need to broken out, preferably as a 2.54mm pinheader (3 or 4 pin)
	- SWCLK
	- SWD
	- GND
	- 3.3V (optional, but usefull if available)

[image]

4. (optional) Supply +5V to the other split half instead of +3.3V

bstiq, 2022-07-11:
More things to do : 

- [ ] remove on/off button
- [ ] fix gpio2 text (not oriented correctly)
- [ ] add more gpio headers

bstiq, 2022-08-09:
Issue 3 is done.
Issue 2: At the moment I won't support it.
Issue 1: I am creating a separate issue for it

closed 2022-08-09 by bstiq