Bastardkb/Dilemma #8: Beta test feedback

keyboard-magpie, 2022-08-02:
Not an issue but thought this would be appropriate place to add feedback, in no particular order:

1. Build guide- suggest add steps for case mounting- was a new technique for me
2. FFC cable steps- I couldn't get it to sit quite neatly with your steps- needed one more fold

More to follow as I finish the build, if i didn't document these fresh I'd for get them!

keyboard-magpie, 2022-08-02:
3. I'm aware it's specified on the github page, but it's probably worth adding specifics about the cirque resistors to the build guide (i.e. for the dilemma assembled DO NOT alter any of the resistors).

KarlK90, 2022-08-05:
1. Mouse bites needed sanding to fit properly in the cases, this could either be a post-process step at bastardkb or be done by the user. Later should probably be documented.
2. Cases tend to tip over on the upper end when used with MX switches and high profile keycaps. Another bumper around the location shown in the picture would be great:
[image]
3. Another "high profile" track-pad assembly for usage with MX switches and high profile keycaps would be awesome. In this combination the trackpad sits quite recessed.

ykz89, 2022-08-05:
1. Not sure if this is already addressed, but with the low profile case, there is no room for the protruding button of the power switch on the right half. I have the assembled version, so I desoldered the power switches anyway.
[image]

2. The reset button is kinda hard to reach, and I feel like it needs a screw here; when pressing the reset button, you're pushing the pcb out of the case.
[image]

0xcharly, 2022-08-07:
Built: Dilemma Assembled Beta tester edition
Switches: Choc v1
Case: low profile

The overall build experience is great. I went out of my way to sand the mouse bites and clean the flux, but otherwise a very fast board to assemble, on par with more classic keyboards assembly.

## Points that have been mentioned already

1. Add a list of tools required at the top of the build guide. That will allow people to get ready ahead of time and not get stuck during assembly: in addition to Torx T-10H, one need some way to sand out the mouse bites (with either a file or sandpaper).

1. The case is in the way of the reset button. It is still accessible, but partially obstructed by the case.

1. The battery features are confusing. The board is not made for wireless, so the on/off switch (which needs to be de-assembled), and the 2 battery connector footprints are surprising. Removing those footprints would give enough space to move the reset button further down (see previous point).

1. The extra half screw hole on the bottom left looks like an oversight and a production issue, in particular because it is not used for assembly. I would either remove it, or make it whole and use it for assembly.

## Other notes

1. While the PCB does not pop out of the case, it feels like a couple screws are missing to properly secure it. At least in the bottom left and top right (for the left side, and top left/bottom right on the right side).

1. The low profile case is missing half a cm of tolerance for the USB-C port cutout: on the right side, it looks like the USB-C port is pushing against the cut out, so the PCB is not sitting perfectly flush (although, it's pretty close) and can pop out of the case in that corner more easily than on the left side

1. The ribbon cable assembly is **MUCH** better than with i2c, but I don't feel very comfortable having to twist the ribbon (which makes assembly and re-assembly a bit dangerous). Flipping either the trackpad or the ribbon connector on the PCB should allow for keeping the ribbon flat/not-twisted.

1. The left side feels "empty", like it's missing something. In particular with all the footprints available on the board, it feels incomplete. On the other hand, seeing the controller exposed is pretty cool IMO. Maybe some sort of 3D printed holder and/or see-through acrylic piece would complete it nicely, and provide a better visual and structural balance with the trackpad side.

1. One advantage of this keyboard over more complicated dacman builds is that you can test the PCB early. I'd recommend including a "Test the PCB" step in the guide by using either Via or Vial (the downside of Vial is that you need to "unlock" the PCB, which requires both halves and serial working, and 2 pairs of tweezers, but the downside of Via is that you need to side-load the JSON, unless we go through the process of upstreaming that part – I heard Via opened up more of it's source code and made contributing new JSON easier). For both tool, we can now point to the web-hosted version from the build guide.

bstiq, 2022-08-08:
Thank you so much everyone !

I created issues based on your feedback.

@ykz89 you will notice there's nothing about the reset button, that's because it's fixed.

@0xcharly , the power stuff is also removed

closed 2022-08-08 by bstiq

bstiq, 2022-09-16:
@0xcharly happy to let you know that the 1.2 RC fixes the cable issue ! I will do a few full build with them to test everything out then deploy on the next batch.
Thanks for the feedback !

0xcharly, 2022-09-16:
Awesome, thanks for addressing this! Can't wait to try it out.